MarlinFirmware/Configurations #1005: [BUG] Artillery Sidewinder X2 Controller Fan not working

Schalonsus, 2024-01-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With stock firmware the controller fan starts spinning when heat bed is over 50°C or stepper motors are working
With bugfix-2.1.x there is only a clicking sound from the controller fan, never turns on
With 2.1.2.1 the controller fan starts spinning over 50°C but then stalls and just make clicking sound

I lowered #define CONTROLLERFAN_SPEED_ACTIVE from 255 to 240 and then the controller fan is working like expected

### Expected behavior

Heat bed over 50°C or stepper motor working the controller fan starts spinning

### Actual behavior

Controller fan does not start spinning only clicking sounds

### Version of Marlin Firmware

2.1.2.1

### Printer model

Artillery Sidewinder X2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/13990526/config.zip)

thisiskeithb, 2024-01-19:
> I lowered #define CONTROLLERFAN_SPEED_ACTIVE from 255 to 240 and then the controller fan is working like expected

Interesting. 255 is max speed, so it should work better than 240.

robherc, 2024-01-19:
Is there any chance his logic levels for the fan got inverted?

Schalonsus, 2024-01-20:
With 2.1.2.1 it starts spinning full throttle and then stalls, so logic should not be inverted

Schalonsus, 2025-08-31:
Just do 240 or anything lower than 255, then it works fine

thinkyhead, 2025-10-03:
Is this true for every Artillery Sidewinder X2 out there? Even so, it is a hardware issue, and it sounds like the settings we provide are able to make it work within its parameters. Is there something actionable for the firmware or configs here? We have these settings available…
```cpp
//#define FAN_MIN_PWM 50
//#define FAN_MAX_PWM 128
```