MarlinFirmware/Configurations #1015: skr mini v3 Define DIAG_JUMPERS_REMOVED

akash1769, 2024-02-10:
using ender 5 pro/CrealityV1 example config. message says 

Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"

There is no sensorless homing defined in this example configuration.h and just adding 

#Define DIAG_JUMPERS_REMOVED

under //#define SENSORLESS_PROBING (dunno if there's a better place) returns

In file included from c:\users\akash\desktop\marlin 2.12\marlin\src\inc\MarlinConfigPre.h:39,
                 from c:\users\akash\desktop\marlin 2.12\marlin\src\inc\marlinconfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\akash\desktop\marlin 2.12\marlin\configuration.h:1417:2: error: invalid preprocessing directive #Define
 1417 | #Define DIAG_JUMPERS_REMOVED
      |  ^~~~~~
Error: Failed to parse Marlin features. See previous error messages.
========================================================== [FAILED] Took 2.11 seconds ==========================================================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:00:02.105

Any suggestions? Google turned up nothing definitive

thisiskeithb, 2024-02-10:
You need to format it as:

```c
#define DIAG_JUMPERS_REMOVED
```

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-02-10 by thisiskeithb

akash1769, 2024-02-10:
Is it just the capital D?

akash1769, 2024-02-10:
Apparently... thanks for your help... I feel silly... I should know better