MarlinFirmware/Configurations #1016: [BUG] Marlin 2.1.2.2 Experimental_Scurve compilation errors with SKR mini E3 v2.0 for Ender 3 config files

altadeos, 2024-02-10:
### Bug Description

I have error messages "error: #error "EXPERIMENTAL_SCURVE is no longer needed and should be removed."" when I try to compile Marlin 2.1.2.2 using Creality's ender 3 SKR mini E3 v2.0 files and compilation fail.

Compilation is OK if I comment "#define EXPERIMENTAL_SCURVE" line 2124 in Configuration_adv.h

Could you please check if this option is still needed and correct the config file ?

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

- Download Marlin 2.1.2.2 source code
- Download Configuration files for 2.1.2.2
- Replace the files in Marlin source directory and try to compile with Marlin Autobuild using Visual Studio Code
[BigTreeTech SKR Mini E3 2.0.zip](https://github.com/MarlinFirmware/Configurations/files/14230428/BigTreeTech.SKR.Mini.E3.2.0.zip)
- Compilation fail
- Comment "#define EXPERIMENTAL_SCURVE" line 2124 in Configuration_adv.h

- Rebuild
- Compilation is OK

**Expected behavior:**

Compilation success.

**Actual behavior:**

Compilation failure.
[image]
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

altadeos, 2024-02-10:
I found that EXPERIMENTAL_SCURVE has been removed in the changelog 
[image]
https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.2.2

thisiskeithb, 2024-02-10:
As the error states, you need to remove `EXPERIMENTAL_SCURVE` from your config.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-02-10 by thisiskeithb