MarlinFirmware/Configurations #1027: marlin will not acknowledge the robin nano v3.1

adowg67, 2024-03-02:
so I try to compile marlin but VS code "build" says I do not have a motherboard even thought it is listed in boards.h AND #define in Configuration.h.  Im sotrry , but this isnt fun anymore/

ellensp, 2024-03-02:
This is not a support channel

This Issue Queue is for Marlin Configuration bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2024-03-02:
In saying that the most common cause is user error

You should have #define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3_1

not

#define BOARD_MKS_ROBIN_NANO_V3_1 
or
#define MOTHERBOARD MKS_ROBIN_NANO_V3_1

And sometimes marlin can say no motherboard after a whole lot of other errors.  You need to look at the first error first

closed 2024-03-02 by ellensp