MarlinFirmware/Configurations #1033: [BUG] Longer LK5 disable ENDSTOP_INTERRUPTS_FEATURE

ellensp, 2024-03-23:
### Bug Description

Longer LK5 needs ENDSTOP_INTERRUPTS_FEATURE disabled, but older versions have it enabled.

✅ bugfix-2.1.x
❌ release-2.1.2.2
❌ release-2.1.2.1
❌ release-2.1.2

Example does not exist further back (that I could find quickly) 

# Need to update these older Config files. 
 
### Steps to Reproduce

1. Download faulty release configs example 
2. Build
3. Get the error "Z_MIN_PIN is not interrupt-capable. Disable ENDSTOP_INTERRUPTS_FEATURE to continue."

**Expected behavior:**

Examples should build

**Actual behavior:**

error "Z_MIN_PIN is not interrupt-capable. Disable ENDSTOP_INTERRUPTS_FEATURE to continue."

thisiskeithb, 2024-03-23:
There are **so** many broken configs in older releases.

Do you really want open that can of worms? 🙃

ellensp, 2024-03-23:
[image]