MarlinFirmware/Configurations #1051: Can you make a version for the: Creality Ender 3 V2 Neo - BTT SKR Mini E3 V3, But with the stock 4.3" Ender 3 V2 Neo LCD Screen?

BIG-BRO-L, 2024-05-05:
Can you make a version for the: Creality Ender 3 V2 Neo - BTT SKR Mini E3 V3, But with the stock 4.3" Ender 3 V2 Neo LCD Screen?

_Originally posted by @BIG-BRO-L in https://github.com/MarlinFirmware/Configurations/issues/1032#issuecomment-2094979429_

thisiskeithb, 2024-05-05:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/649

closed 2024-05-05 by thisiskeithb

BIG-BRO-L, 2024-05-06:
Would you be willing to help me? I'm kind of a noob to coding and this is my first time compiling firmware. I looked through the pin (#649) and can't find the specific marlin build version I'm looking for:

Printer: 
Ender 3 V2 Neo

Motherboard: 
Bigtreetech skr mini e3 v3

Display: 
Stock 4.3" LCD Screen

Software version: 
Marlin 2.1.x

LeviLaswell, 2024-05-06:
> Duplicate of #649
@thisiskeithb 
Where is this version for the stock lcd?

LeviLaswell, 2024-05-06:
@BIG-BRO-L 
Did you find the stock lcd version in #649?