MarlinFirmware/Configurations #1061: [BUG] Invalid option GLOBAL_MESH_Z_OFFSET in bugfix 2.1.x branch

ellensp, 2024-06-02:
### Bug Description

Example configs for bugfix-2.1.x all contain the following in Configuration.h

```
  /**
   * Add Z offset (M424 Z) that applies to all moves at the planner level.
   * This Z offset will be automatically set to the middle value with G29.
   */
  //#define GLOBAL_MESH_Z_OFFSET

```

The code from 2020 for this has not yet been merged https://github.com/MarlinFirmware/Marlin/pull/18987

This is causing much confusion in users who try this option, and it doesn't do anything.

### Steps to Reproduce

1. Look at any example config files in bugfix 2.1.x 

**Expected behavior:**

Only valid configuration options are in example Config files

**Actual behavior:**

Invalid configuration options are present.

#### Additional Information

These are also present in import branch, but import branch can have things that are up and coming, but bugfix branch should not.

thisiskeithb, 2024-06-02:
> These are also present in import branch, but import branch can have things that are up and coming, but bugfix branch should not.

Since `import-2.1.x` auto-pushes to `bugfix-2.1.x` via [`deploy.yml`](https://github.com/MarlinFirmware/Configurations/blob/import-2.1.x/.github/workflows/deploy.yml), `bugfix-2.1.x` will always have the same pending config options.

There are quite a few other changes in the configs repo that do not work / are pending in PRs:

<details><summary>expand me for diff:</summary>
<p>

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index f88f5b9..e21184d 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -20,6 +20,8 @@
  *
  */
 #pragma once
+#error "Don't build with import-2.1.x configurations!"
+#error "Use the 'bugfix...' or 'release...' configurations matching your Marlin version."
 
 /**
  * Configuration.h
@@ -112,6 +114,15 @@
 //#define SERIAL_PORT_3 1
 //#define BAUDRATE_3 250000   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE
 
+/**
+ * Select a serial port to communicate with RS485 protocol
+ * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
+ */
+//#define RS485_SERIAL_PORT 1
+#ifdef RS485_SERIAL_PORT
+  //#define RS485_BUS_BUFFER_SIZE 128
+#endif
+
 // Enable the Bluetooth serial interface on AT90USB devices
 //#define BLUETOOTH
 
@@ -1275,10 +1286,16 @@
  *   M204 P    Acceleration
  *   M204 R    Retract Acceleration
  *   M204 T    Travel Acceleration
+ *   M204 I    Angular Acceleration
+ *   M204 J    Angular Travel Acceleration
  */
-#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E acceleration for printing moves
-#define DEFAULT_RETRACT_ACCELERATION  3000    // E acceleration for retracts
-#define DEFAULT_TRAVEL_ACCELERATION   3000    // X, Y, Z acceleration for travel (non printing) moves
+#define DEFAULT_ACCELERATION                  3000  // X, Y, Z ... and E acceleration for printing moves
+#define DEFAULT_RETRACT_ACCELERATION          3000  // E acceleration for retracts
+#define DEFAULT_TRAVEL_ACCELERATION           3000  // X, Y, Z ... acceleration for travel (non printing) moves
+#if ENABLED(AXIS4_ROTATES)
+  #define DEFAULT_ANGULAR_ACCELERATION        3000  // I, J, K acceleration for rotational-only printing moves
+  #define DEFAULT_ANGULAR_TRAVEL_ACCELERATION 3000  // I, J, K acceleration for rotational-only travel (non printing) moves
+#endif
 
 /**
  * Default Jerk limits (mm/s)
@@ -2092,6 +2109,12 @@
   #endif
 
   /**
+   * Add Z offset (M424 Z) that applies to all moves at the planner level.
+   * This Z offset will be automatically set to the middle value with G29.
+   */
+  //#define GLOBAL_MESH_Z_OFFSET
+
+  /**
    * For Cartesian machines, instead of dividing moves on mesh boundaries,
[... 151 lines cut ...]
+ * Variables
+ *
+ * Define a variable from 100-115 with G-code like '#101=19.6'.
+ * A variable can then be used in a G-code expression like 'G0 X[#101+3]'.
+ * See https://gcodetutor.com/cnc-macro-programming/cnc-variables.html
+ */
+//#define GCODE_VARIABLES
+
+/**
  * Support for MeatPack G-code compression (https://github.com/scottmudge/OctoPrint-MeatPack)
  */
 //#define MEATPACK_ON_SERIAL_PORT_1
@@ -3974,6 +4007,7 @@
 #ifdef G0_FEEDRATE
   //#define VARIABLE_G0_FEEDRATE // The G0 feedrate is set by F in G0 motion mode
 #endif
+//#define G0_ANGULAR_FEEDRATE 2700 // (°/min)
 
 /**
  * Startup commands

```

</p>
</details>

ellensp, 2024-06-02:
Import from import branch should be smart enough not to import those sections...

thinkyhead, 2026-09-02:
This is by design, to allow CI tests to run on PRs with new options without needing to modify a PR's CI behavior or create separate maintained copies of `import-2.1.x` for each PR that introduces a new option. It is imperfect and requires careful curation during releases to remove unpublished changes, so perhaps there is some standard hint that a PR can pick up on to change which config branch is fetched for a PR.

### Candidates for special PR hooks:
> `ci-build-tests.yml` : `make tests-single-ci TEST_TARGET=${{ matrix.test-platform }}` executes `buildroot/bin/run_tests`.
  - Could pass the PR number to `tests-single-ci` and ensure it gets passed on to `run_tests`.

> `buildroot/bin/run_tests` : applies the test `config.ini` file and runs the build, invoking `configuration.py` : `apply_config_ini`.
  - Could modify the URL here to append `@dev-12345` branch name based on the PR number.

> `configuration.py` : `apply_config_ini` handles parsing the `config.ini` and calling `fetch_example`.
> `configuration.py` : `fetch_example(url)` assumes `bugfix-2.1.x` unless the URL specifies `@other-branch` at the end.
  - Could look for an associated PR number in the `GITHUB_ENV` or associated with the checked-out branch.

The script would need to look for the PR branch, then fall back to `bugfix-2.1.x` if it's not found, so making this a logical scripting behavior seems necessary. The least disruptive method would be to have `configuration.py` handle everything by examining the environment, so that's what I'll attempt first.

thinkyhead, 2026-09-05:
I have made the change so that when Configuration options are added by a PR these options can be added in a Configurations branch named after the PR, specifically `pr-#####`. When building a test associated with a PR using `mftest` (whether on CI or locally) any fetched configurations will be pulled from this branch, if it exists.

I will apply a commensurate change for the builds on Configurations. For a branch named `pr-#####` the Configurations test builds will pull from that Marlin PR instead of pulling from `bugfix-2.1.x`.

closed 2026-09-05 by thinkyhead