MarlinFirmware/Configurations #1062: [BUG] Need #define BABYSTEP_Z

JJones780, 2024-06-02:
### Bug Description

In order to enable and see Babystepping in 2.1.2.2  BABYSTEP_Z must be defined.

This #define is missing from at least
"config/examples/Creality/Ender-3 Pro/CrealityV422/Configuration_adv.h"
in branches:
import-2.0.x 
2.1.2.3 
and 2.1.2.2 

### Configuration Files
You can run this to fix every configuration in lower directories:

find . -type f -name "Configuration_adv.h" -print0 | xargs -0 sed -i '' -e 's/\/\/#define BABYSTEP_XY/#define BABYSTEP_Z \n  \/\/#define BABYSTEP_XY/g'

or just fix the one you need.

thisiskeithb, 2024-06-02:
This is not a bug.

closed 2024-06-02 by thisiskeithb

thisiskeithb, 2024-06-03:
> My apologies... BABYSTEP_Z is not only NOT a bug but it is not found in the source code anywhere!

That's because `BABYSTEP_Z` is not a real config flag. To babystep Z, you'd enable `BABYSTEPPING`.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server