MarlinFirmware/Configurations #1065: [BUG] release 2.1.2.4 Configs are quite broken

ellensp, 2024-06-17:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2024-06-17:
I found some other minor issues:

"`// Add 'P' parameter to set percentage done, otherwise use Marlin's estimate`" is in some configs, but not others. It should be "`// Add 'P' parameter to set percentage done`"

"`// Add 'R' parameter to set remaining time, otherwise use Marlin's estimate`" is in some configs, but not others. It should be "`// Add 'R' parameter to set remaining time`"

"`// G-code to run on Stop Print (e.g., "G28XY" or "G27")`" is in some configs, but not others. It should be "`// G-code to run on SD Abort Print (e.g., "G28XY" or "G27")`"

---

These 72 configs consistently show up as the problematical ones:

- Artillery/Sidewinder X1/0.9 BMG - E3D V6
- BIQU/B1 - SKR 1.4
- BIQU/B1 - SKR 2
- BIQU/B1 SE
- BIQU/B1 SE Plus
- BIQU/BX
- BIQU/Thunder Standard
- Creality/CR-10/BigTreeTech SKR Mini E3 2.0
- Creality/CR-10 Mini/BigTreeTech SKR Mini E3 2.0
- Creality/CR-10 S5/BigTreeTech SKR Mini E3 v3
- Creality/CR-10S/BigTreeTech SKR Mini E3 2.0
- Creality/CR-6 SE
- Creality/Ender-3/BigTreeTech E3 RRF
- Creality/Ender-3/BigTreeTech SKR E3 Turbo/2-into-1 Hotend
- Creality/Ender-3/BigTreeTech SKR E3 Turbo/Dual Z
- Creality/Ender-3/BigTreeTech SKR E3 Turbo/Single Extruder
- Creality/Ender-3/BigTreeTech SKR E3-DIP V1.1
- Creality/Ender-3/BigTreeTech SKR Mini E3 1.0
- Creality/Ender-3/BigTreeTech SKR Mini E3 1.2
- Creality/Ender-3/BigTreeTech SKR Mini E3 2.0
- Creality/Ender-3/BigTreeTech SKR Mini E3 3.0
- Creality/Ender-3/MKS Robin E3/V1.0
- Creality/Ender-3/MKS Robin E3/V1.1
- Creality/Ender-3 Max
- Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 1.0
- Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 1.2
- Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 2.0
- Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 3.0
- Creality/Ender-3 S1/STM32F4
- Creality/Ender-5/BigTreeTech SKR Mini E3 1.2
- Creality/Ender-5/BigTreeTech SKR Mini E3 2.0
- Creality/Ender-5 Plus/BigTreeTech SKR 3
- Creality/Ender-5 Plus/CrealityV1
- Creality/Ender-5 Pro/BigTreeTech SKR Mini E3 1.2
- delta/FLSUN/Q5-nano_v1
- delta/FLSUN/QQS-Pro
- delta/Geeetech/G2
- delta/Geeetech/G2Pro
- delta/Geeetech/G2S
- delta/Geeetech/G2SPro
- delta/kossel_clear
- delta/Tevo Little Monster
- Elegoo/Neptune 2
- EXP3D/Imprimante multifonction
- FolgerTech/FT-5 R2
- Geeetech/A20
- Geeetech/A20M
- Geeetech/A20T
- Geeetech/A30
- Geeetech/A30M
- Geeetech/A30T
- Geeetech/D200
- Geeetech/E180
- Geeetech/M201
- MakerFarm/Pegasus 12
- Modix/Big60
- Prusa/MK3
- Prusa/MK3S-BigTreeTech-BTT002
- Renkforce/RF100
- Renkforce/RF100v2
- Renkforce/RF100XL
- Simulator
- Sovol/SV-01/BigTreeTech SKR Mini E3 3.0
- Sovol/SV-01/CrealityV22
- Sovol/SV-01/CrealityV221
- Sovol/SV-02
- Sovol/SV-03
- Two Trees/Sapphire Plus/Sapphire Plus V2
- Two Trees/Sapphire Plus/Sapphire Plus V2.1
- Ultimaker/Ultimaker 2+
- Ultimaker/Ultimaker Original Plus (2.1.1)
- Ultimaker/Ultimaker Original Plus DXU

thisiskeithb, 2024-06-17:
I submitted patch, but I know it'll automatically close when `2.1.2.4` is force-pushed :)

- https://github.com/MarlinFirmware/Configurations/pull/1066

thisiskeithb, 2024-06-21:
https://github.com/MarlinFirmware/Configurations/pull/1066 has been merged / https://github.com/MarlinFirmware/Configurations/tree/release-2.1.2.4 has been fixed.

PR #1066 (2024-06-21): [release-2.1.2.4] 🩹 Fix 2.1.2.4 typos
### Description

> [!IMPORTANT]  
> This is a `release-2.1.2.4`-specific patch

Fixes #1065:

- Config version `02010203` => `02010204`
- Remove `EXPERIMENTAL_SCURVE`
- `#elif SDSUPPORT` => `#elif HAS_MEDIA`
- Fix `SET_PROGRESS_PERCENT` comment
- Fix `SET_REMAINING_TIME` comment
- Fix `EVENT_GCODE_SD_ABORT` comment

### Benefits

`2.1.2.4` configs will build.

### Related Issues

- https://github.com/MarlinFirmware/Configurations/issues/1065
closed 2024-06-21 by thisiskeithb