MarlinFirmware/Configurations #1102: [BUG] local variables don't work

MrKuskov, 2024-09-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

this code is not compiled
```
#define B1_USE_BLTOUCH

....

#define X_MIN_ENDSTOP_HIT_STATE LOW
#define X_MAX_ENDSTOP_HIT_STATE HIGH
#define Y_MIN_ENDSTOP_HIT_STATE LOW
#define Y_MAX_ENDSTOP_HIT_STATE HIGH
#define Z_MIN_ENDSTOP_HIT_STATE TERN(B1_USE_BLTOUCH, HIGH, LOW)
#define Z_MAX_ENDSTOP_HIT_STATE HIGH
#define I_MIN_ENDSTOP_HIT_STATE HIGH
#define I_MAX_ENDSTOP_HIT_STATE HIGH
#define J_MIN_ENDSTOP_HIT_STATE HIGH
#define J_MAX_ENDSTOP_HIT_STATE HIGH
#define K_MIN_ENDSTOP_HIT_STATE HIGH
#define K_MAX_ENDSTOP_HIT_STATE HIGH
#define U_MIN_ENDSTOP_HIT_STATE HIGH
#define U_MAX_ENDSTOP_HIT_STATE HIGH
#define V_MIN_ENDSTOP_HIT_STATE HIGH
#define V_MAX_ENDSTOP_HIT_STATE HIGH
#define W_MIN_ENDSTOP_HIT_STATE HIGH
#define W_MAX_ENDSTOP_HIT_STATE HIGH
#define Z_MIN_PROBE_ENDSTOP_HIT_STATE Z_MIN_ENDSTOP_HIT_STATE
```
required
`#define Z_MIN_PROBE_ENDSTOP_HIT_STATE HIGH`
it used to be compiled in this form

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-09-19:
You are mistaken

Make sure you actually save changes to Configuration.h 

With #define B1_USE_BLTOUCH platformio itself shows

[image]

closed 2024-09-19 by ellensp

ellensp, 2024-09-19:
the "TERN(B1_USE_BLTOUCH, HIGH, LOW)"  just says 

If B1_USE_BLTOUCH is defined use HIGH, else use LOW