MarlinFirmware/Configurations #1104: [BUG] The printing area is invalid for most Ender 3 printers by CREALITY.

BrianWalczak, 2024-09-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using the Ender-3 Neo configuration on my printer with the stepper drivers updated, the extruder did not move to the correct locations. The position seemed to be off proportion, almost as if the printer bed size was incorrectly configured.

### Expected behavior

I expected the printer to move in accordingly to my specified GCODE location during the print.

### Actual behavior

The entire print was shifted left, and incorrectly placed. This was easily identified when the printer ended up drawing the test line too far to the left side.

### Steps to Reproduce

1. Download the latest configuration file, for example, for the Ender 3 Neo.
2. Use Visual Studio Code to update these settings and build the firmware on a microSD card.
3. Update your firmware by inserting the microSD card into your Ender 3 Neo.
4. Try to print any model, in which the printer will first draw a simple line on the left hand side before starting (this happens with any model to make sure the filament won't ooze).

### Version of Marlin Firmware

2.1.2.4

### Printer model

Creality Ender-3 Neo

### Electronics

N/A

### LCD/Controller

N/A

### Other add-ons

The hardware was not modified.

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Here is [a download](https://github.com/user-attachments/files/17177172/config.zip) of my configuration (it was already modified to the correct settings), but the incorrect configuration can be found here: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Neo

This issue can be easily resolved by updating the X_BED_SIZE and Y_BED_SIZE to the printable area rather than the actual bed dimensions. In this case, it should be updated from `235` to `220`. This is correct for models such as the Ender-3 V2 (it's already updated to 220), but not the Ender-3 itself (currently at 235).

thisiskeithb, 2024-09-29:
> The printing area is invalid for most Ender 3 printers by CREALITY

If that were true, this would have been updated a long time ago 😄 

>  it should be updated from 235 to 220

Creality lists the Ender-3 series as 220x220, but most Ender-3s can reach the full bed area of 235x235.

There is some variance in the X & Y travel limits due to endstop placement & printer assembly, but that can be adjusted on a per-machine basis.

closed 2024-09-29 by thisiskeithb

BrianWalczak, 2024-10-04:
Got it! Maybe this was just some confusion, after upgrading to Marlin v2.1, I noticed the print test shifted to the left more. Thought this was a bug but maybe it isn't, I'll flash it again and trying to print a model completely. Thanks again!