MarlinFirmware/Configurations #1106: [BUG] (short description) UBL doesn't probe right (X_MAX_POS) side of BED on Ender 5 Plus

puzari-amit, 2024-09-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Stock Ender 5 Plus Printer with no changes / upgrades whatsoever. Downloaded the latest bugfix files and bugfix reloaded confirm for Ender 5 Plus. Made no changes to the code whatsoever.
Out of the box, when UBL is running, the firmware doesn't probe the right side (where X_MAX_POS endstop) is located while probes the rest of the bed just fine.
I googled the issue and I believe what others have suggested - the UBL is not able to reach the point.
I tried with the MESH_INSET option and the issue disappeared when I set it to 45 however, on the left side (X_MIN_POS) it was probing too far off the edge.
Finally, setting MESH_MAX_X to 45 gave the desired result.
Just putting this out here, incase other users are also facing this issue then, the Config files for E5P could be updated with these numbers.

### Bug Timeline

30/09/2024

### Expected behavior

Probes 25/25 points.

### Actual behavior

Probes 20/25 points leaving out the 5 points on the X_MAX side

### Steps to Reproduce

Stock printer
Default bugfix code
Default bugfix config

### Version of Marlin Firmware

Bugfix 2.1.X

### Printer model

Ender 5 Plus

### Electronics

Stock

### LCD/Controller

Stock

### Other add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Same as my slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_E5P.zip](https://github.com/user-attachments/files/17190194/Configuration_E5P.zip)

ellensp, 2024-09-30:
UBL does not probe where it cannot reach, this is by design.

From your configs
`#define NOZZLE_TO_PROBE_OFFSET { -44, -5, -2.8 }
`
so your probe is left of hotend 44mm 

```
#define X_BED_SIZE 358
#define X_MIN_POS  8
#define X_MAX_POS  X_BED_SIZE + X_MIN_POS
```

So your x-max position is 358+8 so 366
With X probe offset at -44 the furthermost point on the bed you can probe is 366 -44 = 322
So you cannot probe from X322 to X366, as the probe physically cannot move there.

puzari-amit, 2024-09-30:
Yes, I understand that and I agree with your analysis. My point here being, the config file for Ender 5 Plus, should have the MESH_MAX_X set to 45 in the config_adv file so that it does probe that column as well.

ellensp, 2024-09-30:
MESH_MAX_X 45 would limit probing to X8 to X45, ie only a thin strip on the left.

puzari-amit, 2024-09-30:
My mistake. I meant "MESH_MAX_X set to _X_BED_SIZE -_ 45"

ellensp, 2024-09-30:
no, UBL will extrapolate the bed leveling data  for the entire bed. that setting would turn off that feature.  You don't want to set MESH_MAX_X like that. please read P3  in 
https://marlinfw.org/docs/gcode/G029-ubl.html

thisiskeithb, 2024-09-30:
Unless I'm missing a requirement for RELOADED, the default leveling type for these machines should be `AUTO_BED_LEVELING_BILINEAR` instead of `AUTO_BED_LEVELING_UBL` since that is how they were shipped & is the expected leveling type for Cura and PrusaSlicer's start scripts.

puzari-amit, 2024-09-30:
MESH_INSET is set to 15 be default. Does it mean this setting also disables the 

> no, UBL will extrapolate the bed leveling data for the entire bed. that setting would turn off that feature. You don't want to set MESH_MAX_X like that. please read P3 in https://marlinfw.org/docs/gcode/G029-ubl.html

MESH_MAX_X, MESH_MAX_Y are set by X/Y_BED_SIZE - MESH_INSET.
MESH_INSET itself is set to 15 be default. Does this not disable the extrapolation you mentioned?
All I am doing is changing the value for MESH_MAX_X from X_BED_SIZE - MESH_INSET (15) to X_BED_SIZE.
So, compared to the original value, I am decreasing the mesh bound at x max side further by 30.

Also, I read the link you shared.
What I'm looking to do is actually be able to probe those point and not to automatically fill in the unpopulated mesh points.

puzari-amit, 2024-10-01:
> Unless I'm missing a requirement for RELOADED, the default leveling type for these machines should be `AUTO_BED_LEVELING_BILINEAR` instead of `AUTO_BED_LEVELING_UBL` since that is how they were shipped & is the expected leveling type for Cura and PrusaSlicer's start scripts.

Ah. Thank you so much for calling this out, but for a different reason than you think.
Yes, indeed my firmware earlier used bilinear however, the recent config file had UBL enabled and thought of sticking with it.

After I fixed the issue of rightmost column not being probed by adjusting the value of Mesh Max X, I noticed another bug.
With the earlier leveling type of bilinear, the topmost rightmost grid was the last point being probed. After that all the check marks on the grid of the display would become blank followed by the mesh values being displayed.

Now, with UBL, the topmost rightmost grid is not the last one being probed, however due to the code (not sure if its a marlin or DGUS issue), once that point is probed, it still clears the checkmarks from all the previously probed points. And then it checks the remaining 2 points and shows the mesh values.

In summary, there is some code somewhere which says to clear the check marks from all points in the grid once the topmost rightmost point is probed. While this was good for bilinear since that was the last point, with UBL, its kinds odd because that is not the last point being probed and it clears the checkmarks in the middle of the leveling process still going on.