MarlinFirmware/Configurations #1135: [BUG] Custom 2.1 firmware displays incorrect position

LilBub, 2025-03-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi

I've asked in a few places and can't seem to get a reply.  

I'm using a Sovol SV04 with custom Marlin 2.1 firmware.  The author has made the source available and moved on to other projects.

Long story short, the position reading is off by a decimal point.  1.2 instead of 12.0 for Z height, etc.

It seems like there should be an easy fix for this?  Could a Marlin guru take a peek at this?

Link to custom Marlin - https://github.com/johncarlson21/SV04-Marlin-2.1.x
Link issue -  https://github.com/johncarlson21/SV04-Marlin-2.1.x/issues/13

More info  https://github.com/johncarlson21/SV04-IDEX-3D-Printer-Mainboard-Source-code

### Bug Timeline

Upon acquiring machine

### Expected behavior

Movements are displayed incorrect - for example a position of 1.2mm when it should be 12mm.

### Actual behavior

Movements are displayed incorrect - for example a position of 1.2mm when it should be 12mm.

### Steps to Reproduce

1) Jog machine anywhere on the XY or Z axis
2) Display is incorrect

[Configuration.zip](https://github.com/user-attachments/files/19508899/Configuration.zip)

### Version of Marlin Firmware

2.1x

### Printer model

Sovol SV04

### Electronics

Stock

### LCD/Controller

Stock

### Other add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Not in front of machine at moment.  Can provide

ThomasToka, 2025-03-28:
This is probably not the right place for this bug report.. But i know this fork, i know how its done to make it work with dgus.. 

The error here is in the screen firmware, so it displays incorrect values.

[image]

where it should be integer digits 3 and decimal digits 1.

[image]

I have prepared a fixed screen firmware for you.

[DWIN_SET_sv04_fixed_positions.zip](https://github.com/user-attachments/files/19510092/DWIN_SET_sv04_fixed_positions.zip)

Give it a try on your own risk. You can always flash back the original firmware.

LilBub, 2025-03-28:
THANK YOU   Many of us in the SV04 group have hoped we could get a fix for this
I'll share this with the group

thisiskeithb, 2025-03-28:
We don't have the resources to support forks of Marlin. You'll need to contact the maintainer of that fork for further support or migrate to upstream Marlin.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2025-03-28 by thisiskeithb

LilBub, 2025-03-28:
Thanks, @thisiskeithb 
(Glad to see you're alive and well)

ThomasToka, 2025-03-29:
> THANK YOU Many of us in the SV04 group have hoped we could get a fix for this I'll share this with the group

works?

LilBub, 2025-03-29:
> > THANK YOU Many of us in the SV04 group have hoped we could get a fix for this I'll share this with the group
> 
> works?

I'm going to try it today!
Is there a way I can direct message you @ThomasToka  for some follow up questions?

thisiskeithb, 2025-03-29:
Please see my message above.