MarlinFirmware/Configurations #1138: [BUG] (short description)STM32F103RCT6 DIY Ender-3/BigTreeTech SKR Mini E3 2.0 does not have any output

zlw235789, 2025-04-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Use examples\Creality\Ender-3\BigTreeTech SKR Mini E3 2.0   example configuration，Auot Build Marlin STM32F103RC_btt(256k)  It can be compiled successfully.
Using uart1 (PA10 PA9), boot0 0, the compiled firmware can be successfully flashed via STM32CubeProgrammer.

But boot0 1, there is no output after restarting, uart1 (PA10 PA9) USART2 (PA2/PA3), 8mhz crystal oscillator seems to have no output。

This is a core board, no screen etc, only one STM32F103RCT6, 3.3V power supply, crystal oscillator, and nothing else。Is it possible to use an internal crystal oscillator?

### Version of Marlin Firmware

Marlin-lts-2.1.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2025-04-03:
The build environment STM32F103RC_btt also sets the firmware bulid offsets

Ie default is
```
[env:STM32F103RC_btt]
extends                     = common_STM32F103RC_variant
build_flags                 = ${common_STM32F103RC_variant.build_flags}
                              -DTIMER_SERVO=TIM5
board_build.offset          = 0x7000
board_upload.offset_address = 0x08007000
```

So this build environment builds the firmware for 0x08007000 as it expects a bootloader to be at 0x08000000-0x08007000

So if you upload this firmware at 0x08000000, it will not run
If you upload this at 0x08007000 it also will not run as you do not have the bootloader.

Install the bootloader from https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/backed_up_original_bootloaders/SKR%20mini%20E3%20V2.0

And upload firmware to 0x08007000

closed 2025-04-03 by ellensp

zlw235789, 2025-04-04:
There are so many types of bootloaders that I don't want to use them，I want to use the bootloader when the chip is produced, this bootloader cannot be modified。
STM32F103RC_creality(256K)     STM32F103RC_btt(256k)  ...
#
# Creality 256K (STM32F103RC)
#
[env:STM32F103RC_creality]
extends = STM32F103Rx_creality
board   = genericSTM32F103RC
board_build.offset          = 0x0000
board_upload.offset_address = 0x08000000
I changed the offset to 0, and after the compilation is successful, the firmware can be flashed through uart1, and the address is 0x08000000 
But it still doesn't work

ellensp, 2025-04-05:
why are you now using STM32F103RC_creality?

you said your using  BTT_SKR_MINI_E3_V2_0  configs 
which uses one of these,, 

env:STM32F103RC_btt 
env:STM32F103RC_btt_USB 
env:STM32F103RC_btt_maple 
env:STM32F103RC_btt_USB_maple 
 
The build environment used must reflect the correct board.
Ie it must match expected crystals, and io pin re mappings must match..

zlw235789, 2025-04-05:
Because there is no output after BTT_SKR_MINI_E3_V2_0 firmware is flashed, I changed it to STM32F103RC_creality test,
Can't different bootloaders be universal? I can't tell if there's a problem with the bootloader, so I want to use the bootloader from the factory.

The core board seems to be fine, I flashed the firmware of klipper and it works normally, and the klipper bootloader offset is set to 0 ,no bootloader

thisiskeithb, 2025-04-05:
You're misunderstanding the difference between a bootloader & Marlin firmware binary used to flash your motherboard that does not have a bootloader.

If you have an SKR Mini E3 V2.0 with a stock Ender-3 LCD (`CR10_STOCKDISPLAY`), then you'd use the referenced config/firmware. If you do not have that hardware, you would not use that config/firmware.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server