MarlinFirmware/Configurations #1139: [BUG] Labels are not showing with DACAI Ender 3v2 and SRK Mini 3v3

redfox-mx, 2025-04-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Currently labels are not showing in DACAI LCD display (with SKR mini 3 v3.0)

### Bug Timeline

I only have tested in v2.1.2.5 and bugfix-2.1.x

### Expected behavior

Labels to be show xD

### Actual behavior

Main Menu
[image]

LCD Version
[image]

note: while images are from v2.1.2.5 bugfix-2.1.x has the same behaviour

### Steps to Reproduce

1. Download prebuld firmware from Marlinbuilds (2.1.2.5)
2. Install firmware
3. Load configuration
4. Copy private folder from Marlin configurations
5. Install LCD assets into LCD DACAI

### Version of Marlin Firmware

2.1.2.5

### Printer model

Ender 3v2

### Electronics

Motherboard: SKR MINI 3 v3.0

### LCD/Controller

DACAI

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2025-04-06:
Code and asset files must match, you cannot mix them.
Did you use assets from https://github.com/MarlinFirmware/Configurations/tree/2.1.2.5/config/examples/Creality/Ender-3%20V2/LCD%20Files ?

redfox-mx, 2025-04-06:
Yes! Images show behaviour with only marlin assets. When i was trying to solve the issue I test mix both assets.

I try with 2.1.2.5 (firmware and assets) then with  bugfix-2.1.x (firmware and assets). The problem seems to be #807. Before its changes labels are showing correctly. 

PS: #886 solve the problem, but labels does not fit correctly, and then in https://github.com/MarlinFirmware/Configurations/commit/05bdd06f396dbc576adecdc9822b17aff78879e1 assets are breaked again :C