MarlinFirmware/Configurations #1142: [BUG] preview/cache files in LCD Assets of Ender 3 V2

ThomasToka, 2025-04-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

https://github.com/MarlinFirmware/Configurations/archive/refs/tags/2.1.3-b2.zip

in Configurations-2.1.3-b2.zip\Configurations-2.1.3-b2\config\examples\Creality\Ender-3 V2\LCD Files - private.tar.gz

are 

_. files. preview/cache file created by macos or windows.

They need to be removed. I dont know if other LCD assets are also affected.

Dacai displays with private will error on the font.

[image]

[image]

### Expected behavior

LCD assets install.

### Actual behavior

LCD assets error out. See picture

### Steps to Reproduce

1. Load the mentioned private file.
2. Try to install on the display
3. See it error out on the font.

### Version of Marlin Firmware

probably more than one

### Printer model

Creality Ender 3 Max Neo / V2

### LCD/Controller

Dacai

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [x] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Zip file with configs not needed.

thisiskeithb, 2025-04-20:
I can confirm that all of these tarballs have macOS system files in them:

- `/config/examples/AnyCubic/Vyper/LCD files/English_Chinese/DWIN_SET.tar.gz`
- `/config/examples/Creality/Ender-3 V2/LCD Files/DWIN_SET.tar.gz`
- `/config/examples/Creality/Ender-3 V2/LCD Files/private.tar.gz`
- `/config/examples/Creality/Ender-5 Plus/DGUS-Reloaded/DWIN_SET.tar.gz`

thisiskeithb, 2025-04-20:
See https://github.com/MarlinFirmware/Configurations/pull/1143

closed 2025-04-20 by thinkyhead